hanyazou/BMI160-Arduino #2: Correct pinout

abouillot, 2017-04-28:
I'm trying to use a BMI160 on a GY-BMI160 breakout board. The naming convention on this board are quite funny, but I'm puzzled on the correct wiring.

The PIN are labeled as VIN, 3v3, GND - these are obvious.
SCL, SDA for I2C communication, but not supported by your library, then CS and SCK, also obvious in SPI context. 
SAO and SDX and OCS are a bit of question. the two INT1 and INT2 are for the interrupt support.

Do you have any idea of the proper connection?

I'm trying this library on Arduino UNO and ESP8266 WeMos pro mini.

hanyazou, 2017-04-29:
I'm not sure but according to BMI160 Datasheet (http://www.mouser.com/ds/2/783/BST-BMI160-DS000-07-786474.pdf) 4.Pin-out and Connection diagrams, I2C and SPI share 2 pins, clock and data. And SPI needs 2 more pins, MISO and CS.

name |  I2C  |    SPI
----------|----------|----------
SDO  |   -     |     MISO
CSB  |   -     |      CS
SCx   |   SCL  |   SCK
SDx   |   SDA  |   MOSI
OCSB  | -       |    -
OSDO |  -      |     -

You already found the CS pin. So, the SAO pin might be the SDO/MISO pin. (I couldn't find 'SAO' pin in the datasheet.) You can try this pin assign.

Or you could buy another one which is same pinout as mine.
http://www.ebay.co.uk/itm/New-Bosch-6DOF-Inertial-measurement-sensor-Attitude-module-BMI160-/142109967598

hanyazou, 2017-04-29:
BTW, I'm going to add I2C support to my library this long weekend.

abouillot, 2017-04-29:
Thanks for your feedback. I will try further. Is your code as found in this repo working fine with Arduino Uno? I saw somewhere on internet someone has issue using it. I'm happy to fix the connection issue, but I'll need to be sure it has some chance of working :-).

hanyazou, 2017-04-29:
Fortunately, I had that board.
http://pic.twitter.com/mwRW8eajwr

My tester tell me that:

board silk | BMI160 pin | SPI
-------------|-------------|-------------|
SAO | SDO | MISO
CS    | CSB | CS
SDA  | SDx | MOSI
SCL  | SCx  | SCK

I didn't use this board yet. It's brandnew.  I will test this pin assinment tommorrow.

> Is your code as found in this repo working fine with Arduino Uno?

Yes, if I remember correctly. I've checked my library on UNO and 101 at that time. We can fix the library if it does not work.

hanyazou, 2017-04-30:
I found the description of pin names of the board. The names come from STMicro's LSM6DS3. Because LSM6DS3 and BMI160 are very similar PIN assign and comaptible at package size, that board can be used for LSM6DS3 also.

The LSM6DS3 breakout board and BMI160 breakout board which I own are very similar. I think they are same completely.
https://twitter.com/hanyazou/status/858556142636507137/photo/1

LSM6DS3's Application note says:

> pin #1 "SDO/SA0" SPI 4-wire interface serial data output (SDO), I2C least significant bit of the device address (SA0)

Please refer these PDFs for more information.

Datasheet
http://www.mouser.com/ds/2/389/lsm6ds3us-973707.pdf

Application note
http://www.st.com/resource/en/application_note/dm00157511.pdf

hanyazou, 2017-05-01:
I've updated the README.md including SPI and I2C pin assignment of the GY-BMI160 breakout board.

abouillot, 2017-05-08:
Thanks for your work and update.

I've tried again, with the latest code. I've setup the connexion in I2C, commented the init line in SPI mode uncommented the line dor the I2C  mode.
When runnig on ESP8266 or Arduino Uno - without any other change - I still get communication error:

> Initializing IMU device...
BMI160GenClass::i2c_init(): I2C failed.
Wire.endTransmission() failed.
Wire.endTransmission() failed.
Wire.endTransmission() failed.

I can see the device is properly attached, the i2c_scanner code (from arduino.cc) find a device at 0x69.

hanyazou, 2017-05-08:
The I2C default address is 0x68, not 0x68. I guess you leave SAO pin of the sensor breakout board open. Please connect the SAO pin to GND. Or you can specify I2C address in the begin() method.

abouillot, 2017-05-08:
Ok, made some progress. I've wired the SAO and the initialization sequence start.

I had to comment the line mapping the interrupt in the BMI160 initialize method: 
`reg_write(BMI160_RA_INT_MAP_0,  0xFF);`
as otherwise, the program stuck there.

Surprisingly, even a 
`reg_write(BMI160_RA_INT_MAP_0,  0x0);`
block the program.

hanyazou, 2017-05-09:
so, you could retrieve some sensor values from BMI160 w/o any blocking?

abouillot, 2017-05-09:
With disabling the BMI160_RA_INT_MAP_O write, I see data coming in. I have to check them, as I see a lot of variations, even when the sensor stay fix on the table.

closed 2019-02-16 by hanyazou

juliusjob324, 2024-08-04:
MINE HAS A  I2C device found at address 0x6A !........GIVING wire end transmission error