hanyazou/BMI160-Arduino #3: do you know how to read the analog data by MatLab?

hooowardyin, 2017-05-08:


hooowardyin, 2017-05-08:
in order to read the analog data, which 3 pins on the shuttle board should be connected to A0-A6?

hanyazou, 2017-05-08:
I don't know, sorry. I've googled "shuttle board" and found 
 http://www.mouser.com/ds/2/783/BMI160_Shuttleboard_Flyer-786498.pdf . Are you using this board?
According to the PDF above, there are no analog pin on the board. You should connect it to I2C or SPI socket of Ardunino instead.

hooowardyin, 2017-05-15:
yes, it is this shuttle board. 
Thanks to your code and instruction, I have made the SPI connection to Arduinon UNO and read the data out. 
My next work is to analysis these data in MatLab. MatLab needs to know the exact Arduino pin number(like A1, D5 something) to read the data out. However I can't figure out which pin of the Arduino is used to read data. I think there should be 3 pins to read the angular acceleration from x,y,z directions, but there is no 3 connections between D pin or A pin on Arduino. That makes me confused. Do you know which pins of Arduino are used to read data?

hanyazou, 2017-05-15:
I have no experience with MatLab.  I've googled again and found these functions.

arduinosetup:
https://mathworks.com/help/supportpkg/arduinoio/ug/configure-setup-for-arduino-hardware.html

arduino:
https://mathworks.com/help/supportpkg/arduinoio/ref/arduino.html

If you use these functions to communicate with Arduino, it seems to be difficult to use BMI-160. You should use serial communication or offline analysis.

I found these examples.

serial communication:
http://publications.lib.chalmers.se/records/fulltext/155199.pdf

offline:
http://ardadv.blogspot.jp/2012/05/using-l3g4200d-with-arduino.html

hooowardyin, 2017-05-31:
thank you sir !
I figured out how to read the pin.
My next step is to read the output data out , such as gx, gy and gz.
How can I find the address of gx for example? 
Sorry for these stupid question. I am new to Arduino...

hanyazou, 2017-06-03:
I can't understand your question. How could you read 'the pin'? What is 'the pin'? What is 'the address of gx'???

closed 2019-02-16 by hanyazou