hanyazou/BMI160-Arduino #5: do you know how to connect BMI160 by 3-wire mode?

kocmocgim, 2017-06-29:
I don't understand how connect hardware wire and which registr have to read. (i read the datasheet).
Thank you.

hanyazou, 2017-07-02:
I don't try the 3-wire mode. But the datasheet says that:

3.2Primary Interface
Table 25: Mapping of the primary interface pins
Pin #1 SDO is DNC (float) in SPI 3-wire mode.

3.2.2Primary SPI Interface 
Two  configurations  of  the  SPI  interfaceare  supported  by  the  BMI160:  4-wire  and  3-wire.  The same  protocol  is  used  by  both  configurations.  The  device  operates  in  4-wire  configuration  by default.  It  can  be  switched  to  3-wire  configuration  by  writing   ́1 ́  to Register  (0x6B)  IF_CONFspi3. Pin SDI is used as the common data pin in 3-wire configuration.

I'm afraid that this library does not support the 3-wire mode. So you must modify the library source code. Why don't you use the 4-wire mode instead?

kocmocgim, 2017-07-02:
> Why don't you use the 4-wire mode instead?
I develop a motion capture glove based on 18 BMI160 (furher i would neded for 45 BMI160). So wire reducing will be better for me :)
But a haven't much time to library modify right now :(

uzhul38, 2017-07-13:
Hello, 
I understand that you are trying to use a lot of BMI160 in a row but, if i understood the SPI protocol correctly, the way to collect datas from one of the 18 BMI160 is with specifying it SS pin. SO between 3-wire and 4-wire mode for 18 BMI160 it will change only one wire. Because the MISO/MOSI and SCK pin will be common for all sensors.
So you can use the 4-wire mode of this lib without problems.

kocmocgim, 2017-07-13:
I use 4-wire mode. And it works perfect. But I want to reduce wire lenth. It will improve electroscheme. Different in 1 wire is a high for this case, but not a critical.

Siggi1988, 2020-05-13:
Hello, 

how can I set the sda, scl Pin configuration to use the BMI160 test example on a teensy 3.6?