hanyazou/BMI160-Arduino #13: Misisng eiaextensions.h

Lukas-Sturm, 2020-11-18:
Hi, this might be a stupid question.

I am using PlatformIO and when I try to build. I am missing the `eiaextensions.h`
In the Arduino Web IDE it builds as expected.
I found that the eiaextensions.h comes from some Drivers for the Curie board.
https://github.com/arduino/ArduinoCore-arc32
https://github.com/arduino/ArduinoCore-arc32/blob/master/system/libarc32_arduino101/drivers/eiaextensions.h

Is this library supposed to work for Arduino ?
And what does the `eiaextensions.h` do? It seems to be some binding to some external function (with all the external definitions) ?

bastian2001, 2021-06-28:
Hi Lukas, have you found a solution? Haven't really found another working library for the BMI160 over SPI so far and it seems like I'm too stupid to do the SPI stuff myself.

Lukas-Sturm, 2021-06-29:
Sadly not
I just ended up looking at the source code and copied the bits I needed.
You can also look at the Datasheet it's also quite helpfull.

Hope that helps you :)

bastian2001, 2021-06-29:
Thanks for your reply, I forked this library and fixed the issues for me. I don't know if a pull request is correct with the limited testing I've done. Also, I just copied and pasted files, some might be obsolete, idk. But it works :)
https://github.com/bastian2001/BMI160-Arduino-1