hanyazou/BMI160-Arduino #15: I2C Initialization

Satientas58, 2021-03-31:
Dear Sir/Madam,

I have utilized the code and patched it according to the configuration. However, I am unable to initialize the I2C using "BMI160.begin(BMI160GenClass::I2C_MODE, i2c_addr);".

May I check whether there is a solution to this?

Thank you.

Kanken6174, 2021-12-05:
I have the exact same issue, the module will not initialise on I2C or SPI, it gets stuck on init call

Kanken6174, 2021-12-05:
Actually this was solved by using PlatformIO instead of the arduino IDE, no idea what happend there 🤷

mtruma, 2022-05-17:
Hi, I have the same problem on arduino and on platformio I still get stuck on initialisation.

mtruma, 2022-05-17:
Whoever finds this problem here is how I solved it:
add at the beginning of **void setup()**
**Wire.begin();** 

and function 
**BMI160.begin(BMI160GenClass::I2C_MODE, i2c_addr);**
should be
**BMI160.begin(BMI160GenClass::I2C_MODE, Wire, i2c_addr);**
second argument in function should be **Wire**