hanyazou/BMI160-Arduino #19: Wire.endTransmission() Failed when trying to use I2C

Calferino, 2023-11-10:
I am using the default code but getting Wire.endTransmission() failed. I tried the other suggestions in the issues such as including Wire.begin but new problems arise with  BMI160.begin(BMI160GenClass::I2C_MODE, Wire, i2c_addr);

YouthfulPat501, 2024-01-31:
having the same issue here. can't figure it out.

Mboudali, 2024-02-12:
Hi there, 
Looks like an I2C address issue.
Run an I2C scanner (google search) and check your sensor's address. 
Use the following line to declare your sensor:
`BMI160.begin(BMI160GenClass::I2C_MODE, Wire, 0x69);`

Replace 0x69 with the address you found using the I2C scanner.

kittenarmy, 2024-07-01:
Trying to use Wire as above wouldn't compile; cannot find function for twowire&. find_i2c example return 0x69 for my module.

This worked for me (Arduino 1.8.9, Nano):

'''
#include <BMI160Gen.h>

// use pin 14 for interrupt, doesn't seem to be needed 
const int select_pin = 10;
const int i2c_addr = 0x69;

void setup() {
  Serial.begin(9600); // initialize Serial communication
  while (!Serial);    // wait for the serial port to open

  // initialize device
  Serial.println("Initializing IMU device...");
  BMI160.begin(BMI160GenClass::I2C_MODE, i2c_addr);

     // Set the accelerometer range to 250 degrees/second
  BMI160.setGyroRange(250);
  Serial.println("Initializing IMU device...done.");
}

void loop() {
  int gxRaw, gyRaw, gzRaw;         // raw gyro values
  float gx, gy, gz;

  // read raw gyro measurements from device
  BMI160.readGyro(gxRaw, gyRaw, gzRaw);

  // convert the raw gyro data to degrees/second
  gx = convertRawGyro(gxRaw);
  gy = convertRawGyro(gyRaw);
  gz = convertRawGyro(gzRaw);

  // display tab-separated gyro x/y/z values
  Serial.print("g:\t");
  Serial.print(gx);
  Serial.print("\t");
  Serial.print(gy);
  Serial.print("\t");
  Serial.print(gz);
  Serial.println();

  delay(500);
}

float convertRawGyro(int gRaw) {
  // since we are using 250 degrees/seconds range
  // -250 maps to a raw value of -32768
  // +250 maps to a raw value of 32767

  float g = (gRaw * 250.0) / 32768.0;

  return g;
}'''

kittenarmy, 2024-07-01:
Trying to use Wire as above wouldn't compile; cannot find function for twowire&. find_i2c example return 0x69 for my module.

This worked for me (Arduino 1.8.9, Nano):

```
#include <BMI160Gen.h>

// use pin 14 for interrupt
const int select_pin = 10;
const int i2c_addr = 0x69;

void setup() {
  Serial.begin(9600); // initialize Serial communication
  while (!Serial);    // wait for the serial port to open

  // initialize device
  Serial.println("Initializing IMU device...");
  BMI160.begin(BMI160GenClass::I2C_MODE, i2c_addr);

   // Set the accelerometer range to 250 degrees/second
  BMI160.setGyroRange(250);
  Serial.println("Initializing IMU device...done.");
}

void loop() {
  int gxRaw, gyRaw, gzRaw;         // raw gyro values
  float gx, gy, gz;

  // read raw gyro measurements from device
  BMI160.readGyro(gxRaw, gyRaw, gzRaw);

  // convert the raw gyro data to degrees/second
  gx = convertRawGyro(gxRaw);
  gy = convertRawGyro(gyRaw);
  gz = convertRawGyro(gzRaw);

  // display tab-separated gyro x/y/z values
  Serial.print("g:\t");
  Serial.print(gx);
  Serial.print("\t");
  Serial.print(gy);
  Serial.print("\t");
  Serial.print(gz);
  Serial.println();

  delay(500);
}

float convertRawGyro(int gRaw) {
  // since we are using 250 degrees/seconds range
  // -250 maps to a raw value of -32768
  // +250 maps to a raw value of 32767

  float g = (gRaw * 250.0) / 32768.0;

  return g;
}
```